erikarenhill/solaredge-modbus-hass #1: Add support for reading RS485 modbus meters through the inverter

erikarenhill, 2020-02-12:
To get values such as own consumption, grid power etc from the external modbus meter it should be possible to get those values from the inverter as well according to the [SolarEdge SunSpec standard](https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf)

I suggest a small prototype to see if it works with 1st meter.

Meters base address:
* 1st meter – 40000 + 121
* 2nd meter – 40000 + 295
* 3rd meter – 40000 + 469

awulf, 2020-02-15:
I have got it working with meter 1, attached is the modified Sensor.py file
[sensor.txt](https://github.com/erikarenhill/solaredge-modbus-hass/files/4208371/sensor.txt)

erikarenhill, 2020-02-15:
> I have got it working with meter 1, attached is the modified Sensor.py file
> [sensor.txt](https://github.com/erikarenhill/solaredge-modbus-hass/files/4208371/sensor.txt)

Thats great! Would you mind doing a pull request for it? Otherwise I can copy paste it.

awulf, 2020-02-15:
Probably not in a state where it's good enough for a merge request. There needs to be some way to disable meter 1, or potentially have the config specify the quantity of meters between 0 - 3. If you go to addresses where there are no meters it seems to cause errors, I haven't figured out how to pass in configurations yet.

erikarenhill, 2020-02-15:
> Probably not in a state where it's good enough for a merge request. There needs to be some way to disable meter 1, or potentially have the config specify the quantity of meters between 0 - 3. If you [...]

Ok, i can have a look. I might consider splitting the meter to a separate sensor in the same component. Ill let you know when there is code available for testing that out!

erikarenhill, 2020-02-15:
@awulf  please see the code in the branch named feat/meter1 where I copied your code and made it configurable as another sensor. You need to add "read_meter1: true" to configuration.yaml under the solaredge block where your ip/port is.

Please let me know if it works, as I can not test myself without owning the hardware

awulf, 2020-02-16:
@erikarenhill Looks good and values are reading correctly, I have made a small pull request with changes to keep the values seperate in their own entities

erikarenhill, 2020-02-22:
Resolved with PR  https://github.com/erikarenhill/solaredge-modbus-hass/pull/3

closed 2020-02-22 by erikarenhill