erikarenhill/solaredge-modbus-hass #4: Add support for S0 meters

dawiduniec, 2020-02-18:
Hello,

Is it possible to get information about consumption from S0 meter?

erikarenhill, 2020-02-22:
Hello @dawiduniec , no it is not. This is dedicated to SolarEdge Inverters. I suppose you are talking about S0 output from electricity meters?

dawiduniec, 2020-02-22:
I have S0 output from elektricity connected to solaredge , it's a cheaper modbus alternative 

https://www.solaredge.com/pl/products/metering-and-sensors/solaredge-s0-meter-adapter-cable#/

erikarenhill, 2020-02-23:
> I have S0 output from elektricity connected to solaredge , it's a cheaper modbus alternative
> 
> https://www.solaredge.com/pl/products/metering-and-sensors/solaredge-s0-meter-adapter-cable#/

Aha! Thanks for the link. I have to read up on this, didnt know it was possible

erikarenhill, 2020-02-23:
@dawiduniec if you add the meter1 to the lovelace ui and then click the sensor to see all the attributes, does the attribute exported or imported attributes contain any values?

dawiduniec, 2020-02-25:
There is not attributes under meter1

ugoogalizer, 2020-02-26:
Hey Erikarenhill, 

From what I can gather you are coding up the meter and may not have one yourself to test against?

I have a meter and just set the HA "read_meter1" to true, and a whole new entity appeared in HA called "sensor.solaredge_modbus_meter_1" and it's attributes for my device are currently: 

```
ac_current: 9.9
ac_current_phase_a: 0.4
ac_current_phase_b: 8.8
ac_current_phase_c: 0.6
ac_voltage_phase_ll: 422.87
ac_voltage_phase_ab: 422.85
ac_voltage_phase_bc: 420.7
ac_voltage_phase_ca: 425.07
ac_voltage_phase_ln: 245.02
ac_voltage_phase_an: 245.02
ac_voltage_phase_bn: 241.63
ac_voltage_phase_cn: 245.84
ac_frequency: 50.02
ac_power_output: -2209
ac_va: 2317
ac_var: 64837
ac_pf: 81.06
exported: 12721
imported: 3672
exported_va: 0
imported_va: 0
imported_var_q1: 0
imported_var_q2: 0
exported_var_q3: 0
exported_var_q4: 0
events: 0
unit_of_measurement: W
friendly_name: SolarEdge Modbus Meter #1
icon: mdi:power-plug
```
(I have 3 phase power)

So looks like it's functioning but I have not validated the output yet.  If you don't have your own to test against, I'm happy to help.

Also - seriously excellent work on this!

erikarenhill, 2020-02-26:
@ugoogalizer I have a SolarEdge inverter but not any external meter. The implementation for reading the meter 3 phase meter is coming from @awulf that helped out implementing this and verified it working. Glad you found it useful

This ticket is to support the simpler S0 interface (which I don't own either)

ugoogalizer, 2020-02-26:
Ah righteo, sorry for mixing up the tickets but thanks still to @erikarenhill and @awulf for the great work so far!

My inverter is a: P-SE-SE10000H-AU000BWU4 (SolarEdge 10kW HD-Wave 1Ph Inverter) and meter is a: SE-WNC-3Y400-MB-K1.

erikarenhill, 2020-02-26:
> Ah righteo, sorry for mixing up the tickets but thanks still to @erikarenhill and @awulf for the great work so far!
> 
> My inverter is a: P-SE-SE10000H-AU000BWU4 (SolarEdge 10kW HD-Wave 1Ph Inverter) and meter is a: SE-WNC-3Y400-MB-K1.

No worry! Thanks for the feedback :)