erikarenhill/solaredge-modbus-hass #5: Lovelace ui completely unresponsive after start

atze09, 2020-02-20:
Hello erikarenhill,
First of all thank you for your awesome work, this was to only thing i was missing in my solaredge setup.
The local data is a very important topic for my home automation, before i used the api data from the cloud.

Now to my problem:
I installed your integration and added the sample configuration to my configuration.yaml
` solaredge_modbus:
     name: 'SolarEdge Modbus'
     host: 192.168.17.127
     port: 502
     scan_interval: 60 #Optional: time in seconds between readings. Recommended value is at fastest 2-3 sec, default 5
     read_meter1: false #Optional: set this to true if you have an external modbus meter connected with RS485 for reading self consumption`

The inverter shows connected and i can receive data, but since i have this running my lovelace ui is very unresponsive it takes more than 30s to switch pages, so it could not be used as a daily driver.

When i uncomment it and restart it is normal again.
Do you have a hint, what could cause this problem?

Thanks in advance.
Regards,
Dominik

erikarenhill, 2020-02-22:
hi @atze09  I added a sleep for a few millis to the loop in sensor.py in a try to reduce the cpu load, can you try updating to the latest file and see if it works better? Should not be necessary but give it a try if it makes a difference..

atze09, 2020-02-23:
> hi @atze09 I added a sleep for a few millis to the loop in sensor.py in a try to reduce the cpu load, can you try updating to the latest file and see if it works better? Should not be necessary but give it a try if it makes a difference..

Hello Erik,
Thank you for your quick reply, i tried the fix and it works like a charm.
Everything is smooth again. I had a look on the processor sensor and it stayed at ~5% without your component, old version and updated version. So it seems equal but works.

I have an additional question for the external meter (i installed the solaredge version and added it with read_meter1: true) but i cannot read additional values per phase.
Is there a list of readings i can get with and without the external meter?

Once again awesome work!!!

atze09, 2020-02-23:
Is it possible to see all Sensors in Home Assistant?

erikarenhill, 2020-02-23:
> Is it possible to see all Sensors in Home Assistant?

Configuration -> Entities

This component provides two sensors. One for the inverter and one for the meter1. Each sensor provides multiple attributes that can be read

atze09, 2020-02-23:
> > Is it possible to see all Sensors in Home Assistant?
> 
> Configuration -> Entities
> 
> This component provides two sensors. One for the inverter and one for the meter1. Each sensor provides multiple attributes that can be read

I already saw the two Sensors, but where can i See the attributes? The Sensors seems to be not connected, maybe i missed the 2 minutes init of the Inverter.

erikarenhill, 2020-02-23:
@atze09 you can read the source code to see all the attributes set on the meter1 sensor, will add some more docs about this later... but for now see from line 299 in sensor.py and forward and you will see that all phases are added.

https://github.com/erikarenhill/solaredge-modbus-hass/blob/master/custom_components/solaredge_modbus/sensor.py#L299

For example to read ac_current_phase_a into a custom sensor value you should be able to add a template sensor in your configuration.yaml with something like this (see sample-configuration.yaml how to structure the template sensors):

`        solaredge_meter1_ac_current_phase_a:
            friendly_name: "AC Meter 1 Current Phase A"
            value_template: "{{ state_attr('sensor.solaredge_modbus_meter_1', 'ac_current_phase_a') }}"
            unit_of_measurement: 'A'
`

The sample file does not yet contain examples of the meter 1 attributes

Then repeat for every attribute you want to get a custom sensor for. One option you have is to put the data in influx db and then query for the attributes there. Does this help?

erikarenhill, 2020-02-23:
If you setup all attributes as sensors, please copy paste them to me and I can add them to the sample-configuration.yaml file if anyone else wants them!

atze09, 2020-02-23:
Yes thank you, it seems i am offline for eine reason. I need to fix this First. I will keep you updated.

atze09, 2020-02-23:
@erikarenhill 
[image]
Seems the component does not load and my information was wrong. I will investigate.

erikarenhill, 2020-02-23:
> @erikarenhill
> [image]
> Seems the component does not load and my information was wrong. I will investigate.

Sorry thats a sloppy error from my side, I'll fix it immediately

erikarenhill, 2020-02-23:
@atze09 try now and update sensor.py then restart home assistant

atze09, 2020-02-25:
When i try to integrate your component i get this log.
Maybe this causes the unresponsive ui.
`2020-02-25 20:46:10 ERROR (Recorder) [homeassistant.components.recorder.util] Error executing query: (sqlite3.DatabaseError) database disk image is malformed
[SQL: INSERT INTO events (event_type, event_data, origin, time_fired, created, context_id, context_user_id) VALUES (?, ?, ?, ?, ?, ?, ?)]
[parameters: ('platform_discovered', '{"service": "load_platform.sensor", "platform": "solaredge_modbus", "discovered": {"name": "solaredge_modbus", "scan_interval": 5, "read_meter1": false}}', 'LOCAL', '2020-02-25 19:43:40.594176', '2020-02-25 19:46:10.163081', 'f4664a92cd594373ac32ac96ef2a4412', None)]
(Background on this error at: http://sqlalche.me/e/4xp6)
2020-02-25 20:46:10 ERROR (Recorder) [homeassistant.components.recorder] Error saving event: <Event platform_discovered[L]: service=load_platform.sensor, platform=solaredge_modbus, discovered=name=solaredge_modbus, scan_interval=5, read_meter1=False>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1246, in _execute_context
    cursor, statement, parameters, context
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 588, in do_execute
    cursor.execute(statement, parameters)
sqlite3.DatabaseError: database disk image is malformed

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/recorder/__init__.py", line 348, in run
    session.flush()
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 2479, in flush
    self._flush(objects)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 2617, in _flush
    transaction.rollback(_capture_exception=True)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/util/langhelpers.py", line 68, in __exit__
    compat.reraise(exc_type, exc_value, exc_tb)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 153, in reraise
    raise value
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 2577, in _flush
    flush_context.execute()
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/unitofwork.py", line 422, in execute
    rec.execute(self)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/unitofwork.py", line 589, in execute
    uow,
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/persistence.py", line 245, in save_obj
    insert,
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/orm/persistence.py", line 1137, in _emit_insert_statements
    statement, params
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 982, in execute
    return meth(self, multiparams, params)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 293, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1101, in _execute_clauseelement
    distilled_params,
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1250, in _execute_context
    e, statement, parameters, cursor, context
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1476, in _handle_dbapi_exception
    util.raise_from_cause(sqlalchemy_exception, exc_info)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 398, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 152, in reraise
    raise value.with_traceback(tb)
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1246, in _execute_context
    cursor, statement, parameters, context
  File "/usr/local/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 588, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.DatabaseError: (sqlite3.DatabaseError) database disk image is malformed
[SQL: INSERT INTO events (event_type, event_data, origin, time_fired, created, context_id, context_user_id) VALUES (?, ?, ?, ?, ?, ?, ?)]
[parameters: ('platform_discovered', '{"service": "load_platform.sensor", "platform": "solaredge_modbus", "discovered": {"name": "solaredge_modbus", "scan_interval": 5, "read_meter1": false}}', 'LOCAL', '2020-02-25 19:43:40.594176', '2020-02-25 19:46:10.163081', 'f4664a92cd594373ac32ac96ef2a4412', None)]
(Background on this error at: http://sqlalche.me/e/4xp6)`

atze09, 2020-02-25:
At the moment i have this issue.
`2020-02-25 21:08:22 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 4`

erikarenhill, 2020-02-25:
@atze09 that probably means you have a problem communicating with the inverter, try and restart the inverter (power off, on). Also make sure no other clients try to connect as SolarEdge only accept one client

atze09, 2020-02-25:
> @atze09 that probably means you have a problem communicating with the inverter, try and restart the inverter (power off, on). Also make sure no other clients try to connect as SolarEdge only accept one client

I restartet the inverter and now get 4 times the error 2 then again error 4. hm
`2020-02-25 21:33:28 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-25 21:33:28 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-25 21:33:33 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-25 21:33:34 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-25 21:34:09 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 4
2020-02-25 21:34:39 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 4
2020-02-25 21:35:09 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 4
2020-02-25 21:35:40 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 4`

erikarenhill, 2020-02-25:
@atze09 what model of the SolarEdge inverter do you have, and which firmware? 
If you turn off this component, can you fetch information with another modbus tcp client from the inverter?

atze09, 2020-03-03:
> @atze09 what model of the SolarEdge inverter do you have, and which firmware?
> If you turn off this component, can you fetch information with another modbus tcp client from the inverter?

I have the SE15k with display and firmware 3.2.
In the meantime i got it working by playing around a bit with the inverter settings guided by AWulf (Instruction also in the thread in the home assistant forum), this must be a bug of Solaredge.
So now it works like charm and very smoothy, the unresponsive think must be an issue of Modbus communication.

If i have time i will try to help you with the above mentioned sensors.

Thank you for this awesome component and your time!!

closed 2020-03-03 by atze09