erikarenhill/solaredge-modbus-hass #6: Regarding 'utility meter' and storing previous value.

jolaca01, 2020-02-23:
I have detected that there are some issues using the values obtained from the meters whenever there's a glitch (HA restart for example or others) while reading the meter, then if the previous value is not stored, the value imported for the current period could get wrong. This problem has been commented here:

[https://github.com/home-assistant/home-assistant/issues/20931](url)

It would be great if this could be solved as I've already had problems with the import and export values. Thanks a lot!!

jolaca01, 2020-02-25:
Hi these images show the problem I'm having: suddenly the imported or exported values change without apparent reason. 
[image]

[image]

From almost 0kWh (from 0h) to 275kWh ¿?

jolaca01, 2020-02-26:
This issue does not happen with values coming from modbusTCP directly, thus I've never had a problem getting PV solar energy, for example. However, it does happen with data from meter1, like the import or export fields. Just saying in case it may help to solve it. Thanks anyway.

awulf, 2020-03-01:
I have the same thing happening, haven't looked into why it might be. Might also be related to a different issue i'm having which looks like the solaredge-modbus sensor has an error around 1am local time and causes the InfluxDB component to crash.

awulf, 2020-03-04:
So what I think is happening around 1:08AM the Inverter reboots every couple of days, probably got a memory leak problem in the firmware so its been scheduled to do this. I suspect this cause at the same time the router refreshes the DHCP lease for it. So the inverter reboots, and while polling the interface get the errors: 4 = MB_RECV_ERR, 2 = MB_CONNECT_ERR. Do you see these error too in your home assistant logs @jolaca01 ? And judging by the home assistant sensor errors and the influxDB error, some bad values are getting set. So the fix probably is to protect against bad values.

Here is one of the logs:
```2020-02-27 01:08:23 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 4
2020-02-27 01:08:23 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-27 01:08:24 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-27 01:08:24 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-02-27 01:08:55 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
Exception in thread InfluxDB:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/src/homeassistant/homeassistant/components/influxdb/__init__.py", line 370, in run
    count, json = self.get_events_json()
  File "/usr/src/homeassistant/homeassistant/components/influxdb/__init__.py", line 328, in get_events_json
    event_json = self.event_to_json(event)
  File "/usr/src/homeassistant/homeassistant/components/influxdb/__init__.py", line 248, in event_to_json
    json["fields"][key] = float(value)
OverflowError: int too large to convert to float
2020-02-27 01:08:56 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.solaredge_lifetime_production fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 279, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 459, in async_device_update
    await self.async_update()
  File "/usr/src/homeassistant/homeassistant/components/template/sensor.py", line 224, in async_update
    self._state = self._template.async_render()
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 221, in async_render
    return compiled.render(kwargs).strip()
  File "/usr/local/lib/python3.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "/usr/local/lib/python3.7/site-packages/jinja2/_compat.py", line 28, in reraise
    raise value.with_traceback(tb)
  File "<template>", line 1, in top-level template code
OverflowError: int too large to convert to float
2020-02-27 01:12:35 INFO (MainThread) [homeassistant.components.updater] Submitted analytics to Home Assistant servers. Information submitted includes {}
2020-02-27 01:12:35 INFO (MainThread) [homeassistant.components.updater] The latest available version of Home Assistant is 0.105.5```

awulf, 2020-03-05:
Have made a pull request that seems to fix the issue. Have tested by power cycling and all seems good so far but time will tell

jolaca01, 2020-03-05:
Sorry, I've just seen your answer. Yes, I'm seeing exactly the same thing in my home assistant log: 
At the same time, error 2 in my log and this status:
[image]

[image]

jolaca01, 2020-03-05:
@awulf  I'll also try your changes and report back if everything's fine. Thanks a lot!!

jolaca01, 2020-03-09:
@awulf , so far, so good. I didn't have any problem since I installed your changes. I already have my daily and weekly utility meters reporting fine. My monthly meters are still wrong, as they won't reset till next month, other than that, everything's working fine.  Thank you!!

jolaca01, 2020-03-30:
Hi, @awulf  on Saturday I had to restart several times my Raspberry Pi3B+ and then some bad values showed again and they were affecting my utility meters. Is there anyway to revert this back?? Thanks

pueyo5, 2020-04-16:
Same issue here, tried @awulf PR and seems that now is working correctly. I'll keep you updated with any new.

Arjenlodder, 2020-04-27:
Just my two cents; I see 'error 2' in this issue. I also had 'error 2' while trying to get this running, but didn't know what it meant. The constants file of the 'pyModbusTCP' library provided the answer for me: https://github.com/sourceperl/pyModbusTCP/blob/master/pyModbusTCP/constants.py

I simply entered the wrong IP address, so there was no connection. Maybe this helps people.

miezie, 2020-10-24:
There is an extensive thread about missing states in the utility meter component itself. It comes down to bad reading components who do not implement the restore state right. 

See: https://github.com/home-assistant/core/issues/20931

So when this is implemented for this bug... it’s done most probably (and in the right way). Now it is the stated monkey patch ;).

erikarenhill, 2021-03-01:
closing this issue as several people has reported the merged fix from @awulf is working. Thanks!

closed 2021-03-01 by erikarenhill