erikarenhill/solaredge-modbus-hass #8: Add reading Consumption values from SE WYE (4-wire) Meter

kostasyrtariotis, 2020-03-29:
In the Sunspec doc [here](https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf) at page19  the address 40155 handles Export + Import, Production and Consumption, even though there are no related values.

erikarenhill, 2020-06-22:
> In the Sunspec doc [here](https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf) at page19 the address 40155 handles Export + Import, Production and Consumption, even though there are no related values.

Since I don't have a smart meter myself this will require help from someone that has one to verify that it works.

marpi82, 2020-08-05:
Inside 40155 register is stored meter function which is set up at FC setup menu - this is meter function (place of the meter installation)

You can read about this here:
https://www.solaredge.com/sites/default/files/solaredge-meter-installation-guide.pdf
page 29

erikarenhill, 2020-08-05:
Feel free to submit a PR of you are able to test it, even if there is documentation for it I prefer tested code before adding it.