erikarenhill/solaredge-modbus-hass #9: Need to enable Modbus TCP?

raoulteeuwen, 2020-03-30:
Just followed your install instructions (thanks for the code & effort!). But i don't get any data. I also found https://github.com/binsentsu/home-assistant-solaredge-modbus. That page includes a "Enabling Modbus TCP on SolarEdge Inverter" > am i right i also need to do that for your code to come up with data?

dystechnic, 2020-04-09:
Yes you do. 
You can do it the way bintensu shows you in the link you provided.

raoulteeuwen, 2020-04-10:
Maybe stupid question, but after searching pretty thorough, i get the impression wifi on my se8k is optional. And i probably don't have it, so can't configure it, and so can't use Modbus (since it reads Modbus TCP: no option to use it over the existing wired LAN connection?)...

dystechnic, 2020-04-10:
Yes you can use it over the LAN connection.I do to. I own a different inverter than you (SE5000H) and used the internal WiFI accesspoint to configure the inverter and enable modbus over TCP. 

I do not know how you can configure your inverter but I noticed a file on the solaredge site about the se4k-10k [https://www.solaredge.com/sites/default/files/se-three-phase-inverter-setapp-ds-nl.pdf](url). I've got the feeling you are Dutch, so I added the dutch file. It says that it can be commissioned throught  build-in WiFI Accesspoint with the SetApp app. But this can be omitted by using the procedure bintensu discribes on his github page.
I do not know how you can enable the built-in accesspoint, but on my inverter it is through a 3-way switch on the bottom of the inverter. 
 
Maybe this will help you get on track again.

raoulteeuwen, 2020-04-10:
You are right about the nationality ;-). I managed to change some settings and enable ModBus (had to keep-press the button on the bottom to enter a menu, short-press to go to next option etc). But i still get 'None' for all values, solaredge_status reports 0. Any way i can check from a command prompt whether values are published from the Solaredge/ip/port?

raoulteeuwen, 2020-04-10:
I find this in the HA log

" Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:210
First occurred: 7:08:18 PM (100 occurrences)
Last logged: 7:16:34 PM
error 2"

raoulteeuwen, 2020-04-10:
And the log is filling up with

...2020-04-10 19:23:14 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:19 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:24 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:29 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:34 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:39 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:44 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:49 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:54 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:23:59 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:04 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:09 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:14 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:19 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:24 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:29 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:34 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:39 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:44 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:49 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:54 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:24:59 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:25:04 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:25:09 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
2020-04-10 19:25:14 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] error 2
...

dystechnic, 2020-04-11:
Sorry but there is not enough information to debug your config. Can you share more details about your configuratoin? What did you put in configuration.yaml for eample.
Maybe that can help shed some light on the problem.

raoulteeuwen, 2020-04-11:
Thank you for the help. O.k. if you 'give up' on my issue. I used your sample code:

`solaredge_modbus: #https://github.com/erikarenhill/solaredge-modbus-hass
    name: 'SolarEdge Modbus'
    host: 192.168.1.32
    port: 1502
    scan_interval: 5 #Optional: time in seconds between readings. Recommended value is at fastest 2-3 sec, default 5
    read_meter1: false #true #default false #Optional: set this to true if you have an external modbus meter connected with RS485 for reading self consumption`

and 

`  - platform: template #https://github.com/erikarenhill/solaredge-modbus-hass
    sensors:
        solaredge_ac_total_current:
            friendly_name: "AC Total Current"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_total_current') }}"
            unit_of_measurement: 'A'
        solaredge_ac_current_phase_a:
            friendly_name: "AC Current Phase A"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_current_phase_a') }}"
            unit_of_measurement: 'A'
        solaredge_ac_current_phase_b:
            friendly_name: "AC Current Phase B"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_current_phase_b') }}"
            unit_of_measurement: 'A'
        solaredge_ac_current_phase_c:
            friendly_name: "AC Current Phase C"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_current_phase_c') }}"
            unit_of_measurement: 'A'
        solaredge_ac_voltage_phase_ab:
            friendly_name: "SolarEdge AC Output Voltage (A-B)"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_voltage_phase_ab') }}"
            unit_of_measurement: 'V'
        solaredge_ac_current_voltage_a:
            friendly_name: "AC Voltage Phase A"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_voltage_phase_a') }}"
            unit_of_measurement: 'V'
        solaredge_ac_current_voltage_b:
            friendly_name: "AC Voltage Phase B"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_voltage_phase_b') }}"
            unit_of_measurement: 'V'
        solaredge_ac_current_voltage_c:
            friendly_name: "AC Voltage Phase C"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_voltage_phase_c') }}"
            unit_of_measurement: 'V'
        solaredge_ac_power_output:
            friendly_name: "AC Power Output"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_power_output') }}"
            unit_of_measurement: 'W'
        solaredge_dc_current:
            friendly_name: "DC Current"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'dc_current') }}"
            unit_of_measurement: 'A'
        solaredge_dc_voltage:
            friendly_name: "DC Voltage"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'dc_voltage') }}"
            unit_of_measurement: 'V'
        solaredge_dc_power_input:
            friendly_name: "DC Power Input"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'dc_power_input') }}"
            unit_of_measurement: 'W'
        solaredge_heatsink_temp:
            friendly_name: "Inverter Heat sink Temperature"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'heat_sink_temperature') }}"
            unit_of_measurement: '°C'
        solaredge_inverter_efficiency:
            friendly_name: "Inverter efficiency"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'computed_inverter_efficiency') }}"
            unit_of_measurement: '%'
        solaredge_ac_frequency:
            friendly_name: "AC Frequency"
            value_template: "{{ state_attr('sensor.solaredge_modbus', 'ac_frequency') }}"
            unit_of_measurement: 'Hz'
        solaredge_lifetime_production:
            friendly_name: "Lifetime production"
            value_template: "{{ (state_attr('sensor.solaredge_modbus', 'ac_lifetimeproduction') / 1000.0) | round(2) }}"
            unit_of_measurement: 'kWh'
        solaredge_status:
            friendly_name: "SolarEdge Status"
            value_template: >-
              {% if state_attr('sensor.solaredge_modbus', 'status') == 1 %}
                Off
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 2 %}
                Sleep
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 3 %}
                Wake-Up
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 4 %}
                Working
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 5 %}
                Throttled
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 6 %}
                Shut-Down
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 7 %}
                Fault
              {% elif state_attr('sensor.solaredge_modbus', 'status') == 8 %}
                Maintenance
              {% else %}
                Unknown
              {% endif %}`

dystechnic, 2020-04-11:
Hi,
It it not my sample code. I am just a user like you. ;-)
I do not see anything wrong with your configuration. The IP address is the address your inverter has on your network? Can you ping that address for example?
The 'error 2' is probably an eror returned by the pyModbusTCP library.In the Modbus reference it says that erro 2 refers to an 'Illegal Data Address' error. So, maybe the address you are trying to read is wrong for your inverter (does it support the Sunspec protocol?) or the server could not be reached. Hence my IP address question.
An other thing could be the port number. In stead of 1502 you could try 502.That portnumber is used in older versions of the modbus implementations.

Good Luck!

raoulteeuwen, 2020-04-11:
Ah, sorry, thank you :-). Yes, the IP was pingable, i just assumed i had to use the port provided in the sample code, and you were right: for my config, it is 502. I could have known that if i looked at my display better, but you know what they say about assumptions :-|. Works, great, thank you. 

I don't get values for all entities from the template, but that maybe depends on the equipment. Stuff i don't get values for:

- Entity not available: sensor.solaredge_heat_sink_temperature
- sensor.solaredge_computed_inverter_efficiency
- sensor.solaredge_ac_lifetimeproduction
- sensor.solaredge_solaredge_status
- sensor.solaredge_ac_voltage_phase_a (and _b and _c)

closed 2020-04-11 by raoulteeuwen