erikarenhill/solaredge-modbus-hass #13: Negative value of meter#1

magpern, 2020-06-01:
Why is the value negative? How should it be interpreted?
No production is active. It is night time

[image]

pkret, 2020-06-15:
You have probably already figured this out by now, but it means that you are buying/pulling 2.6kW from the grid at that moment. If you are interested in the solar output, see sensor.solaredge_modbus, for how much you are selling to (or buying from) the grid use sensor.solaredge_modbus_meter_1. And difference between both is what your home is using at that moment.
```
        solaredge_usage_now:
            friendly_name: "Usage"
            value_template: "{{ states('sensor.solaredge_modbus')|int - states('sensor.solaredge_modbus_meter_1')|int }}"
            unit_of_measurement: 'W'
```

magpern, 2020-06-15:
> You have probably already figured this out by now

Yes, thanks anyway. I jumped the gun a little too early.

closed 2020-06-15 by magpern

magpern, 2020-06-16:
Any idea of how to get this value as a positive value?

I have reconfigured my external meter to just be a consumption meter. So, for me the value should be positive, meaning "Consumption power usage" instead of "AC power output"

pkret, 2020-06-16:
Create a new template sensor and add *-1... it will then be negative when exporting power to the grid:

```
sensor:
  - platform: template
    sensors:
        solaredge_meter_power_output:
          friendly_name: "Consumption power usage"
          value_template: "{{ state_attr('sensor.solaredge_modbus_meter_1', 'ac_power_output')|int * -1 }}"
          unit_of_measurement: 'W'
```