erikarenhill/solaredge-modbus-hass #15: Battery Sensors

Luckymoep, 2020-07-26:
Hi Guys,

I just got my Solaredge Inverter installed and your custom component is working really great.
One thing I am missing is the status of my connected battery.

From my research I found out, that these information is available in the registers 62836 (current energy flow), 62852 (battery percentage) and 62854 (status)

My python skills are very limited. Would it be possible to integrate this?

mmN83, 2020-12-14:
Hi Luckymoep, 

Can you share document/source where you have found the registers map including the battery status?

Luckymoep, 2020-12-14:
Register 62836 as float(swap) battery power in W
Register 62852 as float(swap) battery status in percent
Register 62854 as uint16 is the status (1=off, 3=charge, 4=discharge, 6=preserve charge)

mmN83, 2020-12-17:
[battery_sensor.zip](https://github.com/erikarenhill/solaredge-modbus-hass/files/5711954/battery_sensor.zip)

Finally something that seems to work. I have not a battery attached to my inverter yet so I kindly ask you if you can do some test on the attached repository.

What I have done:
- Created a new battery sensor inside the solaredge custom component able to read the registers you have provided
- Merged also the second optional meter if you have one (I have only one in my set-up)
- Provided some additional files for your configuration

What you should do to let it work:
- Inside the archive you will find the custom_components folder: copy it under your HA configuration folder or simply add solaredge_modbus subfolder to your existing one.
- Add the code inside configuration.yaml to your configuration.yaml file
- Add sensor.yaml to your configuration directory (the same with configuration.yaml)
- Add solaredge.yaml in the same place as above
- Configure solaredge.yaml with the data of your inverter: IP address, modbus port
- Check the configuration in the server management tab and if ok restart HA server

After that you should be able to access new sensors named:
- sensor.battery_energy_flow
- sensor.percentage
- sensor.status

As in the example.jpg provided below:

[image]

My values are not correct since the battery is not connected but you should see something coherent. 
Let me know if the number read from your set-up are meaninful, maybe I have to tune the data conversion.

Last but not least: do a back before starting :)

mmN83, 2020-12-29:
Did you manage to perform some tests?

Luckymoep, 2020-12-29:
Sorry, christmas and the end of year stuff didn't leave me a lot of time...

When I integrate your code I get a new sensor:
[image]
but the readings are not the same as via api:
[image]

I don't have a second meter too, so I set it to 'false' in configuration.yaml

Luckymoep, 2020-12-29:
Following the discussion in the german forum, it looks like the registers are for an older version.
When I try to poll the registers via modpoll, I get an error:
[image]

But I found an interesting document:
[Power Control Open Protocol for SolarEdge Inverters.pdf](https://github.com/erikarenhill/solaredge-modbus-hass/files/5752547/Power.Control.Open.Protocol.for.SolarEdge.Inverters.pdf)

Which leads me to this:
[image]
Converting for example E140 from hex to decimal is 57664, but I get the illegal data address error too.

mmN83, 2020-12-30:
[image]

This is what I get from my inverter polling on 62836 as starting address. My inverter is an SE6000H (with LCD display).
It correctly reports ModBus connection active by polling on the main topics on LCD screen. It is reported that a lack of ModBus connection within the first 2 minutes put the inverter modbus in idle state so try to switch off and on again while HA is polling on the bus.

By the way, I had found the document in the forum. Maybe it is time to give it a look now :D
By polling at 57734 (battery status) I get the following which reports a status 7 (should mean "idle") that is consistent with the fact I don't have a battery.

[image]

In the next days I will implement the new addresses and I let you know the necessary modification to your files.
Meanwhile let me know if you are able to poll somewhat the inverter via modbus. 

Which is yours? Can you provide the ModBus configuration? 

cheers

Luckymoep, 2020-12-30:
My inverter is a SE10K without LCD Display.
This is a hybrid converter with an integrated storeedge module to connect the BYD LV battery box.

I can poll other registers without problems, so the connection should not be the problem.

In some openhab forum I found the information, that maybe, the storedge is on another slave ID, but I had no luck with that idea either.

Do you have an idea how I can do a full dump of all available registers?

mmN83, 2021-01-01:
Nope, I don't know how to do It.
I saw that your config. should makes use of CAN between battery and storedge interface and this could be the tricky point.
Basically I don't know if the inverter stores/converts the info from CAN bus in some of the rs485 registers.
My inverter allows connection to battery through rs485 (physical) and should report data in dedicated registers.

Could you share the openhab forum link so I can have a look?

Luckymoep, 2021-01-02:
Here is the [Link](https://www.photovoltaikforum.com/thread/139774-wie-errechnet-man-die-pv-produktion-bei-einem-solaredge-wr-aus-den-modbus-werten/?pageNo=2), see the last posted picture.

They have an option which is calles "PrivateAPI" which is not using modbus as far as I could research it.
https://www.openhab.org/addons/bindings/solaredge/

I think you are right, the hybrid models are using Canbus für communication between the two devices.
I talked to Solaredge support and they confirmed that it is not possible the get the values for the battery via modbus for this model :(
I will use the API for the storage level, 15 min seems ok for that purpose.

quinten94b, 2021-01-23:
My apology that I join this issue/request without any contribution.

Just a question, will this work for a Solaredge SE3000H without display, with added storedge unite/LG resu chem 10H battery on the DC side of the inverter.

In Belgium they just abrutply changed the regulations on solar panels and we need to selfconsume as much as we can. For that instant battery data is very welcome instead of the 15min cloud interval times from the official integration.

SebastianG-42, 2021-02-27:
Hi,
at first: thanks for your work
I found out that i am able to read the correct battery energy flow by changing the wordorder to Endian.Little:

`                if reading:
                    data = BinaryPayloadDecoder.fromRegisters(reading, byteorder=Endian.Big, wordorder=Endian.Little)`

I am not shure if this was the right way but nor the values are ok.

It works from the Adress 57716 (E174) and from 62836 (F574) as well.

But the State of Energy and the Status is not working. It is only read one time after starting the server with 0% instead of 50% and 8192 for the status.

I run a SE8K-RWS and a LG-Chem Resu 6.5

mmN83, 2021-02-27:
Hi,

Let me understand, have you implemented what included in the package I released and it went fine by changing the endian type?
I ask since I'm still waiting for my LG battery and I had not the chance to try it by my self.

SE8K-RWS should be a 8KWp inverter with integrated storeedge interface isn't it? the "sesti-x" I mean.

As soon as I receive my battery I will try to solve the issues.

SebastianG-42, 2021-02-27:
Hi,
yes I inclueded the battery_sensor.zip you put in this discussion and after changing the endian type for the battery part I was able read out the charge and discharge power of the battery (E174).
But the state of charge and the status was not working. 
I have not much experience with programming so I was trying to change the adress or the skip_bytes to get the other values working but no luck. 

TheSE8K-RWS is the 8kWp inverter with the direct wired 48V Battery to the inverter without extra boxes. 
So yes this interface is intigrated in the inverter. There is a 7kWp and a 10kWp model aviable as well and I guess very popular today.

Cereal2nd, 2021-04-12:
if you need some debug info i can always help, since yesterday i have the se3000H with an lgchem battery installed.
so if i can gather some debug info let me know

mkrasselt1, 2021-05-14:
I did Update you code a Llttle and managed to get SOC corretcly of my LG Chem 13 Resu - 
I will finetune a little more to get every register and send a pull request later :)

mmN83, 2021-05-14:
Thank you @mkrasselt1. My batteries are installed but still not connected :( I Hope to get back soon on the topic.

SebastianG-42, 2021-05-14:
That sounds very good!

Next thing is to control the charging and discharging of the battery.

mkrasselt1, 2021-05-17:
[image]
This is what i got working, any improvments ?
I noticed with my modbustcp tool that especialy my battery does not provide all the values :/

Remko76, 2021-08-10:
> [image]
> This is what i got working, any improvments ?
> I noticed with my modbustcp tool that especialy my battery does not provide all the values :/
> 
> Edit: Also the Battery Voltage and Amperage is definitly wrong, since my battery is 48V not 800something

Any progress on this? Or do you care sharing your changes? I would like to try that out too....

mkrasselt1, 2021-08-11:
> Any progress on this? Or do you care sharing your changes? I would like to try that out too....

Yes - you can chek it out at my pull request - but current and voltage of battery is still broken - i guess due to them using a DC2DC Converter for low voltage batteries

Remko76, 2021-08-30:
Thanks, I got it all integrated now and getting the right values. I did note that when I took your pull request, is that the "ac_lifetimeproduction" value was completely off. Nothing sensible to make from it. I then changed the Endian to "Big" again and the values are now ok.

blowk, 2022-05-19:
> Thanks, I got it all integrated now and getting the right values. I did note that when I took your pull request, is that the "ac_lifetimeproduction" value was completely off. Nothing sensible to make from it. I then changed the Endian to "Big" again and the values are now ok.

Hi, is it possible to share your code please?

closed 2026-11-09 by erikarenhill