erikarenhill/solaredge-modbus-hass #26: Add scale factor as attribute (or calcutated phase current or even phase power production)

JSdeJong, 2021-04-03:
Thanks for the great addon!
Now I can stop my node-red implementation, which stopped working once in a while.

I do have a request; 
Is it possible to add the scale factor as an attribute, or calculate phase current with the scale factor?
(What I've seen the phase current is the raw data from the modbus).
Or even add the power production per phase

Because I want to store the power usage per phase, I need the power production from the inverter per phase

Thank you very much!

JSdeJong, 2021-04-04:
AC current scale factor is at 40075 register.
I'll check the code and create a pull request if I succeed.

JSdeJong, 2021-04-04:
It seems the registers spit out the wrong scalefactor for the ac current.

ac_current_phase_scalefactor: 0.01
ac_total_current: 2.17
ac_current_phase_a: 0.56
ac_current_phase_b: 0.83
ac_current_phase_c: 0.77
ac_voltage_phase_scalefactor: 0.1
ac_voltage_phase_a: 230.9
ac_voltage_phase_b: 231.9
ac_voltage_phase_c: 233
ac_power_scalefactor: 0.001
ac_power_output: 20

So current of 0.56 with a voltage of 230.9 will result in 129 Watts; however the ac_power_output is only 20 Watts.
But it's not a factor 10 difference; calculating P=U*I for each phase, will result in around 50 Watss instead of the displayed 20.

So it's not going to work what I want. And also it's not an issue caused by the script/custom component but the inverter itself.
Or I'm wrong with my thoughts. 

Any idea?

erikarenhill, 2026-11-09:
As you worked out, the scale factor is already applied to the phase currents. At very low output the inverter's AC current is mostly reactive. 0.56 A at 231 V is about 130 VA apparent power while only ~20 W is real power, which is why P = U·I doesn't match `ac_power_output`. `ac_va`, `ac_var` and `ac_pf` show that difference. Closing since this isn't a bug; feel free to reopen if you see something else.

closed 2026-11-09 by erikarenhill