erikarenhill/solaredge-modbus-hass #27: Lovelace Crash (Browser-Tab & Smartphone-App)

ChristophCaina, 2021-04-05:
Hi,
I am facing regular crashes of the Lovelace Dashboard, when I am on the Tab where I am displaying the SolarEdgeModbus Components.
In my HomeAssistant Logs, the following error occures repeatingly:

``` Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:472
Integration: SolarEdge Modbus (documentation)
First occurred: 14:32:49 (54 occurrences)
Last logged: 14:37:15
error 7

erikarenhill, 2021-04-21:
Hi @ChristophCaina ,

Can you please give some more information:
Version of Home Assistant
What version of Solar Edge inverter you have, and how it's connected (ethernet, wifi)
Any other smart meters connected to the Solar Edge inverter

erikarenhill, 2021-07-26:
closing due to lack of information and can not reproduce it

closed 2021-07-26 by erikarenhill