erikarenhill/solaredge-modbus-hass #33: Setting battery charge/discharge profile

nerfherder, 2021-07-13:
It is possible to control battery charge and discharge profiles by setting values to modbus register 63242 provided register 63236 is set to remote control.

I have set this up using the default modbus integration in hassio but this would be better setup as a default control in this integration.

The registers and settings are defined in https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/

quinten94b, 2022-03-01:
Can you give an example how you implemented it? I just ordered my battery and like to implement this feature also

nerfherder, 2022-03-01:
Have a look at this repo https://github.com/binsentsu/home-assistant-solaredge-modbus it's been implemented there

quinten94b, 2022-03-01:
> Have a look at this repo https://github.com/binsentsu/home-assistant-solaredge-modbus it's been implemented there

thank you. I'm in the progress of migrating to that integration right now.

erikarenhill, 2022-04-07:
Closing this issue as you seem to have found a solution in another integration

closed 2022-04-07 by erikarenhill