erikarenhill/solaredge-modbus-hass #34: Feature request: support for energy module in 2021.8.0

membersound, 2021-08-06:
Is it possible to add the following sensor attributes for support the new energy integration that will come in HA version 2021.8.0?
https://developers.home-assistant.io/blog/2021/05/25/sensor_attributes/

When I look for my solaredge sensors in the new 2021.8.0 energy monitoring, they are not listed and refers to:
https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics

erikarenhill, 2021-12-14:
Hi all, sorry for late reply but it's not a problem using this component with energy-dashboard either. Just add a utility meter:

```
utility_meter:
  solar_production_daily:
    source: sensor.solaredge_lifetime_production
    cycle: daily
```

I will look into adding the state_class to use the sensors directly

kostasyrtariotis, 2022-02-01:
Hi, may I ask how to use the Solaredge meter readings for consumption within the energy dashboard?

erikarenhill, 2026-11-09:
The Energy dashboard works with this integration through a template sensor on the `ac_lifetimeproduction` attribute with `device_class: energy` and `state_class: total_increasing`. Select it, or a utility meter built on it, as the solar production source.

#69 adds this to `sample-configuration.yaml` and documents it in the README, including how to migrate from the old `platform: template` format that newer Home Assistant versions no longer accept. Closing; please reopen if anything is still missing.

closed 2026-11-09 by erikarenhill