erikarenhill/solaredge-modbus-hass #36: Remove last_reset for 2021.09

B-Hartley, 2021-08-31:
in 2021.09 they have changed how measurement sensors work, so last_reset is no longer required.

```
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_ac_energy_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_exported_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_exported_a_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_exported_b_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_exported_c_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_imported_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_imported_a_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_imported_b_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.
2021-08-31 07:09:37 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solaredge_mb_m1_imported_c_kwh (<class 'custom_components.solaredge_modbus.sensor.SolarEdgeSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise report it to the custom component author.

```

erikarenhill, 2021-12-14:
hi @B-Hartley I'm not sure you have opened this issue in the right repository, that log looks like it's coming from another component https://github.com/binsentsu/home-assistant-solaredge-modbus/ am I right?

closed 2021-12-14 by erikarenhill

B-Hartley, 2021-12-14:
Very sorry !