erikarenhill/solaredge-modbus-hass #37: Please, a bit of help installing component

ctrl-alt-d, 2021-09-06:
I followed "manual install" of **[How to install](https://github.com/erikarenhill/solaredge-modbus-hass#how-to-install)** (copy into folder):

[image]

What is the next step?

I tried to add integration via `/config/integrations` but I'm unable to locate this component:

[image]

(Also I tried to restart HA)

Sorry about this noob question.

ternzer, 2021-09-09:
same issue. Repo install/restart HA. Solaredge Modbus is not listed in Configuration>Integrations. Tested another integration just to be safe. Also reinstalled this repo.

seanw18, 2021-09-13:
same same

borgswen, 2021-12-10:
Hi, This component shoud be set up trough yaml code. It's sort of complicated but there are helpfull guides out there.

erikarenhill, 2021-12-14:
Hello,

You have to follow the examples by looking here:
https://github.com/erikarenhill/solaredge-modbus-hass/blob/master/sample-configuration.yaml

It is not configurable through the UI yet.
The first 7 rows is the most important to set up communication with the inverter such as IP and port.

closed 2021-12-14 by erikarenhill