erikarenhill/solaredge-modbus-hass #39: Multiple Inverters

CollabSensei, 2021-12-01:
I have 3 inverters that I am trying to connect via modbus tcp. With this plugin, how do you have multiple ip's?

erikarenhill, 2021-12-14:
that is unfortunately currently not possible

closed 2021-12-14 by erikarenhill

CollabSensei, 2021-12-14:
I ended up creating multiple folders and modifying some of the code, so I could have multiple "instances." Then I called it via something like:
solaredge_modbus1:
  name: 'Inverter 1'
  host: <ip1>
  port: 1502
  scan_interval: 30
  read_meter1: false
  
solaredge_modbus2:
  name: 'Inverter 2'
  host: <ip2>
  port: 1502
  scan_interval: 30
  read_meter1: false
  
solaredge_modbus3:
  name: 'Inverter 3'
  host: <ip3>
  port: 1502
  scan_interval: 30
  read_meter1: false