erikarenhill/solaredge-modbus-hass #40: Error 9

pepsonEL, 2021-12-10:
Hi
I have this error in HA

Ten błąd pochodzi z niestandardowej integracji.

Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:212
Integration: SolarEdge Modbus (documentation)
First occurred: 11:06:23 (66 occurrences)
Last logged: 11:11:53

error 9

erikarenhill, 2021-12-14:
That error is something to do with problems reading from your inverter and not an exception from the component/code itself.

Are you using wifi or ethernet cable to your inverter, are you using the correct IP/port ?
If using wifi, please try to use a ethernet cable instead, that has been proven to work better.

pepsonEL, 2021-12-14:
I use WIFI but inverter works ok on wifi and in solarEdge app all works ok. POrt is ok. But suppoort solar edge update mi firmware and after that i get this error.

erikarenhill, 2021-12-14:
> I use WIFI but inverter works ok on wifi and in solarEdge app all works ok. POrt is ok. But suppoort solar edge update mi firmware and after that i get this error.

Double check that modbus tcp is enabled after the firmware update and verify ip/port

Several other has reported trouble using wifi for modbus tcp, reporting to solaredge servers work a bit different

pepsonEL, 2021-12-14:
I checked it and port 1502 is enable for modbus....

pepsonEL, 2021-12-14:
I installed other integration and it works ok on modbus....
https://github.com/binsentsu/home-assistant-solaredge-modbus

erikarenhill, 2021-12-14:
Ok, then it's not a problem with this specific component if others doesn't work either.

What inverter model do you have, and which version of the firmware?
Have you tried to restart Home Assistant?

I would advise you to try an ethernet cable if the update did something with modbus over wifi. Other users as I said did have problems using wifi with modbus tcp, and that is out of my control

pepsonEL, 2021-12-14:
Yes i restarted many times HA.
My inverter model is: SE5k

CPU Version: 4.14.107
DSP1 Version: 1.20.939
DSP2 Version: 2.20.609
WSA Version: 4.14.70

pepsonEL, 2021-12-14:
SolarEdge block modbus by WIFI. I get this info from other owner integration.

erikarenhill, 2022-04-07:
Will close this as SolarEdge seems to not support modbus tcp over wifi and it's not an issue with the integration itself

closed 2022-04-07 by erikarenhill