erikarenhill/solaredge-modbus-hass #42: 2022.2.2 Detected blocking call inside the event loop. This is causing stability issues.

fb22, 2022-02-06:
Since hass update from 2021.12.10 to 2022.2.2 my log is filled with:

WARNING (MainThread) [homeassistant.util.async_] Detected blocking call inside the event loop. This is causing stability issues. Please report issue to the custom component author for solaredge_modbus doing blocking calls at custom_components/solaredge_modbus/sensor.py, line 75: sleep(0.005)

godfreym29, 2022-02-07:
Also seeing this.

Logged as second part of Issue #43, but HA log details copied here:

Log Details (WARNING)
Logger: homeassistant.util.async_
Source: util/async_.py:144
First occurred: 15:30:51 (2587 occurrences)
Last logged: 17:22:48

Detected blocking call inside the event loop. This is causing stability issues. Please report issue to the custom component author for solaredge_modbus doing blocking calls at custom_components/solaredge_modbus/sensor.py, line 75: sleep(0.005)
Detected blocking call inside the event loop. This is causing stability issues. Please report issue to the custom component author for solaredge_modbus doing blocking calls at custom_components/solaredge_modbus/sensor.py, line 277: sleep(0.005)

march83, 2022-02-24:
3 options.

1. change your log level in configuration.yaml to hide warnings
2. comment out the sleeps in sensor.py (I'm trialling this to see what impact it has on overall stability - would love to hear from the developer what the impact of the sleep is)
3. come up with an alternative to the sleep and make a PR

erikarenhill, 2022-04-07:
I don't recall anymore why there is a sleep in the beginning there, might have been some leftover from an early prototype, so I will just remove it. There is an asyncio.sleep for the duration of polling interval in the end of the loop, no need for two. Will just run it myself for a couple of hours and see that it works as expected first.

erikarenhill, 2022-04-07:
Monitoring cpu on my host running without that sleep I dont see any issues so I'll just push this update right away, don't see how it could cause anything bad