erikarenhill/solaredge-modbus-hass #48: Error: exception: 'int' object is not callable

remspoor, 2022-07-29:
I got this error a few times and am not sure what the impact can be.
Stuff seems to be working fine though.
```
Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:213
Integration: SolarEdge Modbus (documentation)
First occurred: July 27, 2022 at 02:55:30 (8 occurrences)
Last logged: 02:56:29

exception: 'int' object is not callable
```

dverha, 2022-09-15:
Same here since version to 0.1.3.

zeeuk, 2022-12-01:
Have the same

Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:472
Integration: SolarEdge Modbus ([documentation](https://github.com/erikarenhill/solaredge-modbus-hass))
First occurred: 06:01:52 (71 occurrences)
Last logged: 17:07:55

exception: 'int' object is not callable

barrown, 2023-01-26:
Me too

```
ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: 'int' object is not callable
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/solaredge_modbus/sensor.py", line 209, in modbus_loop
    if self._client.last_error() > 0:
TypeError: 'int' object is not callable
```

mrvanes, 2023-06-13:
pyModbus changed the interface of def last_eror() to a property and now has to be called like `self._client.last_error`. I changed that in the code, but then ran into connection problems at pyModus level, so I gave up and created a modbus holding_register count=2 on register  40083 to solve the raw data and scale being out of sync.

Mkay505, 2024-01-08:
Same here. How can this be fixed by a mortal user?

`Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:213
Integration: SolarEdge Modbus ([documentation](https://github.com/erikarenhill/solaredge-modbus-hass))
First occurred: 04:29:11 (2 occurrences)
Last logged: 04:29:44

exception: 'int' object is not callable`

mrvanes, 2024-01-08:
> Same here. How can this be fixed by a mortal user?
>
> `Logger: custom_components.solaredge_modbus.sensor Source: custom_components/solaredge_modbus/sensor.py:213 Integration: SolarEdge Modbus [...]

I quit using the solaredge-modbus-hass integration and now use plain modbus combined with a template to convert the RAW  AC Power value to real Watts.
```
modbus:
  - name: SolarEdge
    type: tcp
    host: 192.168.1.6
    port: 502
    sensors:
      - name: PV AC Power raw
        slave: 1
        address: 40083
        count: 2
        data_type: custom
        structure: ">2h"
        input_type: holding
        #data_type: uint16
        scan_interval: 60
        lazy_error_count: 3
      - name: PV Lifetime Energy
        slave: 1
        address: 40093
        unit_of_measurement: 'kWh'
        input_type: holding
        data_type: uint32
        scan_interval: 60
        lazy_error_count: 3
        device_class: energy
        state_class: total_increasing
        scale: 0.001
        precision: 3
```
and template:
```
    # SolarEdge
  - name: "PV AC Power"
    state: >-
      {% set pv_ac_power_unscaled = states('sensor.pv_ac_power_raw').split(',')[0] %}
      {% set pv_ac_power_scale = states('sensor.pv_ac_power_raw').split(',')[1] %}
      {{ pv_ac_power_unscaled|int(0)*10**pv_ac_power_scale|int(0) }}
    unit_of_measurement: 'W'
    device_class: power
    state_class: measurement
```

barrown, 2024-01-08:
I've switched to https://github.com/WillCodeForCats/solaredge-modbus-multi which is available as a HACS package, and it's been working perfectly ever since.

remspoor, 2024-01-20:
I didn't know this case was still open. I switched to solaredge-modbus-multi as well. so i'll close this case.

closed 2024-01-20 by remspoor