erikarenhill/solaredge-modbus-hass #50: exception: Big

robwolff3, 2023-09-10:
I started getting the following error in home assistant today with the solaredge-modbus-hass integration. Running Home Assistant version 2023.9.1. Any help would be greatly appreciated.

```
2023-09-09 23:10:08.634 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solaredge_modbus
2023-09-09 23:10:08.635 DEBUG (MainThread) [custom_components.solaredge_modbus.sensor] fetching modbus client
2023-09-09 23:10:08.635 DEBUG (MainThread) [custom_components.solaredge_modbus.sensor] creating modbus sensor
2023-09-09 23:10:08.636 DEBUG (MainThread) [custom_components.solaredge_modbus.sensor] added to hass, starting loop
2023-09-09 23:10:08.668 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: Big
```

HejWolf, 2023-09-10:
I receive the same error since update to HA 2023.9.1 yesterday.

HejWolf, 2023-09-10:
Update: After restoring HA 2023.8.3 modbus is working again...

erikarenhill, 2023-09-10:
I'm getting the same, will have a look at it, thanks for reporting

dverha, 2023-09-10:
Same problem here. Resolved by restoring to **core_2023.9.0**

cornubot, 2023-09-10:
Got the same trying to upgrade to 2023.6.1. Rolled back to 2023.5.1

erikarenhill, 2023-09-10:
Apparantly pymodbus was upgraded in 2023.9.1 which caused this as they changed casing of enum Endian.Big to Endian.BIG. Fix is on the way

erikarenhill, 2023-09-10:
please try and update to latest version and see if it fixes your problem, it works for me now. @robwolff3 @HejWolf @dverha @cornubot

closed 2023-09-10 by erikarenhill

robwolff3, 2023-09-10:
The fix works for me. Thank you!!!

dverha, 2023-09-11:
> please try and update to latest version and see if it fixes your problem, it works for me now. @robwolff3 @HejWolf @dverha @cornubot

Works like a charm, thank you!!

PR #51 (2023-09-10): 2023.9.1 fix
Update casing of Endian.Big to Endian.BIG which resolves #50