erikarenhill/solaredge-modbus-hass #52: Not working with 2023.9.1: exception: Big

Timon321, 2023-09-11:
I just reverted from 2023.9.1 to 2023.9.0, because i got following exceptions:

2023-09-11 11:43:19.879 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: Big
2023-09-11 11:43:20.074 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: Big
2023-09-11 11:43:24.897 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: Big
2023-09-11 11:43:25.123 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: Big

There was no data coming from the component.

NdS-Research-Facilities, 2023-09-11:
I got the opposite but I think related error:
` Home Assistant 2023.9.1 Supervisor 2023.08.3 Operating System 10.5 Frontend 20230608.0 - latest `

detail:
```
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/solaredge_modbus/__init__.py:12
Integration: solaredge_modbus (documentation)
First occurred: September 10, 2023 at 19:27:12 (1 occurrences)
Last logged: September 10, 2023 at 19:27:12

Unexpected exception importing component custom_components.solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solaredge_modbus/__init__.py", line 12, in <module>
    from pymodbus.payload import BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

```

seppbusch, 2023-09-13:
same problem here since update to 2023.9.1

2023-09-13 19:02:51.158 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: 'int' object is not callable                                    
Traceback (most recent call last):                                               
  File "/config/custom_components/solaredge_modbus/sensor.py", line 211, in modbus_loop                                                                           
    if self._client.last_error() > 0:                                            
       ^^^^^^^^^^^^^^^^^^^^^^^^^                                                 
TypeError: 'int' object is not callable                                          
                                                                                 
^[c2023-09-13 19:03:26.187 ERROR (MainThread) [custom_components.solaredge_modbus.sensor] exception: 'int' object is not callable                                 
Traceback (most recent call last):                                               
  File "/config/custom_components/solaredge_modbus/sensor.py", line 211, in modbus_loop                                                                           
    if self._client.last_error() > 0:                                            
       ^^^^^^^^^^^^^^^^^^^^^^^^^                                                 
TypeError: 'int' object is not callable                                          
                                        
There was some Modbus Changes in Release notes 2023.9.1 and several posts regarding modes errors.
will try 2023.9.2

seppbusch, 2023-09-13:
2023.9.2 won't work either
i fall back to 2023.9.0 which seems to work again.

NdS-Research-Facilities, 2023-09-13:
Yes, did that also

seppbusch, 2023-09-18:
Had some time for searching in HA forum for Solutions.
Found some hints that 2023.9.0 changed Version of pyModbus. 
https://community.home-assistant.io/t/universal-solar-inverter-over-modbus-rs485-tcp-custom-component-growatt-sofar-solax-solis/140143/1229

There are some other custom modbus Integrations struggeling with Version > 2023.9.0
https://github.com/wills106/homeassistant-solax-modbus/issues/528

@erikarenhill: Any ideas how to proceed ? Wait for a Fix in HA ? or must this be fixed in your Custom Modul ?

erikarenhill, 2023-09-18:
Hello everyone this has already been fixed in the latest version: https://github.com/erikarenhill/solaredge-modbus-hass/pull/51

Please update to version 0.14 in HACS

closed 2023-09-18 by erikarenhill

seppbusch, 2023-09-18:
@erikarenhill: Thanks a lot. Now it is working again.