erikarenhill/solaredge-modbus-hass #53: exception: type object 'Endian' has no attribute 'BIG'

RoyPorter, 2023-10-14:
I've been getting this error for the last couple of weeks, I think I've just tracked down the issue (pymodbus version). I'll check and raise a PR if I'm right.

```
Logger: custom_components.solaredge_modbus.sensor
Source: custom_components/solaredge_modbus/sensor.py:214
Integration: SolarEdge Modbus ([documentation](https://github.com/erikarenhill/solaredge-modbus-hass))
First occurred: 14:30:57 (9 occurrences)
Last logged: 14:31:37

exception: type object 'Endian' has no attribute 'BIG'
```

erikarenhill, 2023-12-10:
This has already been fixed (https://github.com/erikarenhill/solaredge-modbus-hass/issues/50) update to latest version.

closed 2023-12-10 by erikarenhill

RoyPorter, 2023-12-11:
Please re-open, as I experienced this issue when I upgraded to v0.1.4. The linked PR fixes it for me. 

pymodbus v3.5.0 brought in the uppercasing of enum's so it's necessary to pin the version of pymodbus that you're using to one that is at least v.3.5.0.

erikarenhill, 2023-12-11:
Valid point if you are using an old version of Home Assistant

From Home Assistant 2023.9.0 they bumped modbus 3.5.1 to the default, what version of HA are you using?

RoyPorter, 2023-12-11:
Interesting ... I'm on version 2023.11.3

I'm not familiar enough with the inner workings of Home Assistant to workout what's wrong. Pinning the version in solaredge-modbus-hass fixed the issue for me, I'll remove that locally and see if it works with my current version of HA

RoyPorter, 2024-01-08:
So, I'm still experiencing this issue, intermittently. Thought it was because I had another integration also using pymodbus, with a pinned old version, but I've uninstalled and still get this issue. 

I'm on the very latest everything; sometimes it works, but it hasn't wanted to after the latest batches of updates, so I've once again pinned the version in my local install. 

May be an idea to pin it to >= 3.5 ?