erikarenhill/solaredge-modbus-hass #55: HA 2024.1 Breaks Add-On

meremortals70, 2024-01-04:
logger:%20homeassistant.components.template.template_entity%0ASource%3A%20components/template/template_entity.py:194%0AIntegration:%20Template%20(documentation,%20issues)%0AFirst%20occurred:%201:34:10%20PM%20(1%20occurrences)%0ALast%20logged:%201:34:10%20PM%0A%0ATemplateError('TypeError:%20unsupported%20operand%20type(s)%20for%20/:%20'NoneType'%20and%20'float'')%20while%20processing%20template%20'Template%3Ctemplate=(%7B%7B%20(state_attr('sensor.solaredge_modbus',%20'ac_lifetimeproduction')%20/%201000.0)%20%7C%20round(2)%20%7D%7D)%20renders=4%3E'%20for%20attribute%20'_attr_native_value'%20in%20entity%20'sensor.solaredge_lifetime_production'

logger:%20homeassistant.helpers.event%0ASource%3A%20helpers/template.py:569%0AFirst%20occurred:%201:34:10%20PM%20(1%20occurrences)%0ALast%20logged:%201:34:10%20PM%0A%0AError%20while%20processing%20template:%20Template%3Ctemplate=(%7B%7B%20(state_attr('sensor.solaredge_modbus',%20'ac_lifetimeproduction')%20/%201000.0)%20%7C%20round(2)%20%7D%7D)%20renders=2%3E%0ATraceback%20(most%20recent%20call%20last):%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/helpers/template.py%22,%20line%20567,%20in%20async_render%0A%20%20%20%20render_result%20=%20_render_with_context(self.template,%20compiled,%20**kwargs)%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/helpers/template.py%22,%20line%202305,%20in%20_render_with_context%0A%20%20%20%20return%20template.render(**kwargs)%0A%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/local/lib/python3.11/site-packages/jinja2/environment.py%22,%20line%201301,%20in%20render%0A%20%20%20%20self.environment.handle_exception()%0A%20%20File%20%22/usr/local/lib/python3.11/site-packages/jinja2/environment.py%22,%20line%20936,%20in%20handle_exception%0A%20%20%20%20raise%20rewrite_traceback_stack(source=source)%0A%20%20File%20%22%3Ctemplate%3E%22,%20line%201,%20in%20top-level%20template%20code%0ATypeError:%20unsupported%20operand%20type(s)%20for%20/:%20'NoneType'%20and%20'float'%0A%0AThe%20above%20exception%20was%20the%20direct%20cause%20of%20the%20following%20exception:%0A%0ATraceback%20(most%20recent%20call%20last):%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/helpers/template.py%22,%20line%20693,%20in%20async_render_to_info%0A%20%20%20%20render_info._result%20=%20self.async_render(%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/helpers/template.py%22,%20line%20569,%20in%20async_render%0A%20%20%20%20raise%20TemplateError(err)%20from%20err%0Ahomeassistant.exceptions.TemplateError:%20TypeError:%20unsupported%20operand%20type(s)%20for%20/:%20'NoneType'%20and%20'float'%0A

logger:%20homeassistant.setup%0ASource%3A%20setup.py%3A332%0AFirst%20occurred%3A%201%3A33%3A16%20PM%20(1%20occurrences)%0ALast%20logged%3A%201%3A33%3A16%20PM%0A%0AError%20during%20setup%20of%20component%20solaredge_modbus%0ATraceback%20(most%20recent%20call%20last)%3A%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/setup.py%22,%20line%20332,%20in%20_async_setup_component%0A%20%20%20%20result%20=%20await%20task%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/config/custom_components/solaredge_modbus/__init__.py%22,%20line%2036,%20in%20async_setup%0A%20%20%20%20client%20=%20ModbusClient(host,%20port=port,%20unit_id=1,%20auto_open=True)%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/local/lib/python3.11/site-packages/pyModbusTCP/client.py%22,%20line%20116,%20in%20__init__%0A%20%20%20%20self.host%20=%20host%0A%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/usr/local/lib/python3.11/site-packages/pyModbusTCP/client.py%22,%20line%20177,%20in%20host%0A%20%20%20%20raise%20TypeError('host%20must%20be%20a%20str')%0ATypeError:%20host%20must%20be%20a%20str%0A

Jochen754, 2024-01-04:
@meremortals70 
Also got this issue. Got it working again.

Change in the file "init.py" the name "host" to "str(host":

Change:
client = ModbusClient(host, port=port, unit_id=1, auto_open=True)

To:
client = ModbusClient(str(host), port=port, unit_id=1, auto_open=True)

closed 2024-01-05 by erikarenhill

erikarenhill, 2024-01-05:
@meremortals70  update to new version through HACS and it should work again

PR #58 (2024-01-05): Update __init__.py --> Fix for Homeassistant Error in 2024.1.0 --> "TypeError: host must be a str"
Fix Error with Homeassistant Version 2024.1.0: "TypeError: host must be a str"
Fixes: #55, #56, #57