erikarenhill/solaredge-modbus-hass #56: TypeError: host must be a str

membersound, 2024-01-04:
```
Core 2024.1.0
Supervisor 2023.12.0
Operating System 11.2
Frontend 20240103.3
```

```
Logger: homeassistant.setup
Source: setup.py:332
First occurred: 9:12:20 AM (1 occurrences)
Last logged: 9:12:20 AM

Error during setup of component solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 332, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 36, in async_setup
    client = ModbusClient(host, port=port, unit_id=1, auto_open=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyModbusTCP/client.py", line 116, in __init__
    self.host = host
    ^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyModbusTCP/client.py", line 177, in host
    raise TypeError('host must be a str')
TypeError: host must be a str
```

Jochen754, 2024-01-04:
> ```
> Core 2024.1.0
> Supervisor 2023.12.0
> Operating System 11.2
> Frontend 20240103.3
> ```
>
> ```
> Logger: homeassistant.setup
> Source: setup.py:332
> First occurred: 9:12:20 AM (1 occurrences)
> Last logged: [...]

Also got this issue. Got it working again.

Change in the file "init.py" the name "host" to "str(host":

Change:
client = ModbusClient(host, port=port, unit_id=1, auto_open=True)

To:
client = ModbusClient(str(host), port=port, unit_id=1, auto_open=True)

erikarenhill, 2024-01-05:
resolved with #58 update to new version through HACS and it should work again

closed 2024-01-05 by erikarenhill