erikarenhill/solaredge-modbus-hass #57: Component won't worrk with core version 2024.1.0

kostasyrtariotis, 2024-01-04:
log:

Error during setup of component solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 332, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 36, in async_setup
    client = ModbusClient(host, port=port, unit_id=1, auto_open=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyModbusTCP/client.py", line 116, in __init__
    self.host = host
    ^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyModbusTCP/client.py", line 177, in host
    raise TypeError('host must be a str')
TypeError: host must be a str

robwolff3, 2024-01-04:
I am also experiencing this issue.

Jochen754, 2024-01-04:
Also got this issue.

erikarenhill, 2024-01-05:
thanks @Jochen754 for the PR, it has been merged and version is bumped, everyone that has problems, please try and update from HACS and it should work again

closed 2024-01-05 by erikarenhill

Mkay505, 2024-01-05:
Solved:   I have copied the 3 file contents manually from the directory [custom_components/solaredge_modbus]

erikarenhill, 2024-01-05:
@Mkay505 latest version is 0.1.5, sorry I forgot to create the tag/release in github (only bumped the version in the manifest), should show up now!

superharaldsson, 2024-01-05:
Get this error message after i update to 0.1.5 via hacs
Home assistant Core 2024.1.0

Logger: homeassistant.setup
Source: setup.py:332
First occurred: 12:44:56 (1 occurrences)

Last logged: 12:44:56

Error during setup of component solaredge_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 332, in _async_setup_component
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/solaredge_modbus/__init__.py", line 33, in async_setup
    host = conf[CONF_HOST]
           ~~~~^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable