erikarenhill/solaredge-modbus-hass #61: Integration prevents Home Assistent front end from functioning.

user34756361233, 2024-10-20:
Last week (during my holiday) I suddely was not able to access the UI anymore. When arriving home the only way I could get an UI to function was via safe mode. It showed that if I removed solaredge-modbus-hass the problem was gone.

Ref: https://community.home-assistant.io/t/only-userinterface-within-safe-mode/784453

Config:
```
---
# 
# A custom platform component/sensor for reading SolarEdge Modbus TCP into Home Assistant
# consumptions of various utilities (e.g., energy, gas, water, heating).
#
# https://github.com/erikarenhill/solaredge-modbus-hass
# installed via HACS
#
solaredge_modbus:
  name: 'SolarEdge Modbus'
  host: 172.16.0.11 #172.16.0.12
  port: 1502
  scan_interval: 5 #Optional: time in seconds between readings. Recommended value is at fastest 2-3 sec, default 5
  read_meter1: true #Optional: set this to true if you have an external modbus meter connected with RS485 for reading self consumption
```

error log:
```
2024-10-19 23:45:44.363 ERROR (MainThread) [homeassistant.core] Error running job: <Job listen * HassJobType.Callback functools.partial(<function _forward_events_unconditional at 0x7ff059868860>, <bound method WebSocketHandler._send_message of <WebSocketHandler closing=False authenticated=True description=Supervisor from 172.30.32.2 (HomeAssistantSupervisor/2024.10.2 aiohttp/3.10.10 Python/3.12)>>, b'655')>
OverflowError: int too big to convert

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/json.py", line 75, in json_encoder_default
    if hasattr(obj, "json_fragment"):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/functools.py", line 995, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in json_fragment
    return json_fragment(self.as_dict_json)
                         ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/functools.py", line 995, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1884, in as_dict_json
    return json_bytes(self._as_dict)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Integer exceeds 64-bit range

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1562, in async_fire_internal
    self._hass.async_run_hass_job(job, event)
  File "/usr/src/homeassistant/homeassistant/core.py", line 937, in async_run_hass_job
    hassjob.target(*args)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 135, in _forward_events_unconditional
    send_message(messages.cached_event_message(message_id_as_bytes, event))
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/messages.py", line 123, in cached_event_message
    _partial_cached_event_message(event)[:-1],
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/messages.py", line 139, in _partial_cached_event_message
    _message_to_json_bytes_or_none({"type": "event", "event": event.json_fragment})
                                                              ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/functools.py", line 995, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1386, in json_fragment
    return json_fragment(json_bytes(self._as_dict))
                         ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Type is not JSON serializable: State
2024-10-19 23:45:44.654 ERROR (MainThread) [aiohttp.server] Error handling request
OverflowError: int too big to convert

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 477, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 559, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 75, in handle
    result = handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/api/__init__.py", line 228, in get
    body=b"".join((b"[", b",".join(states), b"]")),
                         ^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/api/__init__.py", line 219, in <genexpr>
    states = (state.as_dict_json for state in hass.states.async_all())
              ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/functools.py", line 995, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1884, in as_dict_json
    return json_bytes(self._as_dict)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Integer exceeds 64-bit range

```

membersound, 2025-03-13:
I also had this today and had to deactivate the modbus integration.
Are there alternatives? Seems this is not maintaned anymore

K-Raymer, 2025-03-28:
I'm now having the same issue. Did you find an alterative?

Yorik-Code, 2025-07-24:
Working on from @K-Raymer comment above. I did this also to restore it working.. Things were slightly different in my instance, so I note that here.

You can manually edit the file "sensor.py" located in the custom directory for this integration (for me that is /home/user/Hass/custom_components/solaredge_modbus/sensor.py)

Change line 391 (I assume it was typo above as mine shows line 391)
from 
`m1_energy_scalefactor = 10**data.decode_16bit_uint()`
to 
`m1_energy_scalefactor = 10**data.decode_16bit_int()`

This forces register 40095 to be processed by the "data.decode_16bit_int()" function. Which you can read in payload.py if you are keenly interested.

I guess a pull request could be made for this update if the maintainer is actively accepting pull requests.

erikarenhill, 2026-11-09:
Fixed in [v0.3.1](https://github.com/erikarenhill/solaredge-modbus-hass/releases/tag/v0.3.1). Thanks @cubic-ruddles and @Yorik-Code for finding the signed/unsigned workaround; that was the right lead.

The meter energy scale factor is a signed value but was read as unsigned, so -1 became 65535 and the integration stored `10**65535` in its attributes. Home Assistant can't serialize a number that large (`Integer exceeds 64-bit range`), so the frontend couldn't load states. It only affected `read_meter1: true` with a meter reporting a negative scale factor.

v0.3.1 reads all scale factors as signed, and Modbus reads no longer block Home Assistant's event loop, so a slow or unreachable inverter can't freeze it either. If you applied the manual edit to `sensor.py`, updating through HACS replaces it. Please reopen if you still see this on v0.3.1.

closed 2026-11-09 by erikarenhill