erikarenhill/solaredge-modbus-hass #62: BinaryPayloadDecoder is deprecated

GenMurdock, 2025-02-06:
BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"

GenMurdock, 2025-02-06:
This issue started with the Core upgrade from 2025.1.4 το 2025.2.0

arjanhs, 2025-02-06:
Seeing the same message now

ocsele, 2025-02-17:
me too, throws thousands of entries in the log, quite worried if it will break completely one day. When I tried to manually modify the sensor.py to the expected method (have to admit, with chatgpt help and  without actual coding knowledge :) ), I started receiving the following error message:

`exception: 'ModbusClient' object has no attribute 'convert_from_registers'` 

What's strange is that I still received SOME register values, but not all.

sithmein, 2025-02-18:
Other integrations have the same issue and simply copied the corresponding Python file into their code, e.g. https://github.com/wills106/homeassistant-solax-modbus/pull/1232.

Jochen754, 2025-07-19:
@erikarenhill Is this project still maintained?
I updated to Homeassistant 2025.7 and get this warning.

@sithmein by applying your changes my integration will not start.

Might take a look at the code when I got time.

erikarenhill, 2025-07-19:
@Jochen754 hey, sure it is, but only maintained to work. I have not updated to 2025.07.x myself yet due to vacation and other work but I'll have a look right away! If it breaks I'll fix it. Thanks for reminding me

closed 2025-07-19 by erikarenhill