erikarenhill/solaredge-modbus-hass #64: Modbus not loading after 2025.9 Core upgrade

GenMurdock, 2025-09-04:
After the upgrade of HASS Core to 2025.9 Solaredge-Modbus_hass will not load anymore and gives and error.

Jochen754, 2025-09-04:
Got the same error.

Logger: homeassistant.setup
Quelle: setup.py:506
Erstmals aufgetreten: 17:01:14 (1 Vorkommnis)
Zuletzt protokolliert: 17:01:14

Unable to prepare setup for platform 'solaredge_modbus.sensor': Platform not found (cannot import name 'Endian' from 'pymodbus.constants' (/usr/local/lib/python3.13/site-packages/pymodbus/constants.py)).

Logger: homeassistant.util.loop
Quelle: util/loop.py:77
Erstmals aufgetreten: 17:01:14 (1 Vorkommnis)
Zuletzt protokolliert: 17:01:14

Detected blocking call to import_module with args ('custom_components.solaredge_modbus.sensor',) in /usr/src/homeassistant/homeassistant/loader.py, line 1292: return importlib.import_module(f"{self.pkg_path}.{platform_name}") inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 156, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 712, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 683, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2050, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 199, in _async_component_platform_discovered await self.async_setup_platform(platform, {}, info) File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 324, in async_setup_platform platform = await async_prepare_setup_platform( File "/usr/src/homeassistant/homeassistant/setup.py", line 547, in async_prepare_setup_platform platform = await integration.async_get_platform(domain) File "/usr/src/homeassistant/homeassistant/loader.py", line 1096, in async_get_platform platforms = await self.async_get_platforms((platform_name,)) File "/usr/src/homeassistant/homeassistant/loader.py", line 1161, in async_get_platforms platforms.update(self._load_platforms(platform_names)) File "/usr/src/homeassistant/homeassistant/loader.py", line 1086, in _load_platforms platform_name: self._load_platform(platform_name) File "/usr/src/homeassistant/homeassistant/loader.py", line 1260, in _load_platform cache[full_name] = self._import_platform(platform_name) File "/usr/src/homeassistant/homeassistant/loader.py", line 1292, in _import_platform return importlib.import_module(f"{self.pkg_path}.{platform_name}")

K-Raymer, 2025-09-04:
Bump pymodbus to v3.11.0 removed Endian from constants.

Endian could be manually added, but that's a temporary fix. 

This will require a pretty big rewrite to work work with the latest version of pymodbus

Yorik-Code, 2025-09-05:
I guess that is it then for this one.
This was so heavily baked into my instance as well.

Jochen754, 2025-09-05:
I always update to the new Hass versions straight away.

If it will break I will fix it again.

Yorik-Code, 2025-09-18:
I am not sure why my notifications aren't working as I only just got this now. @Jochen754 thank you very much, it is working for me now also.

I am about to install a larger inverter, 2 new solaredge batteries and more panels. Will see if this custom_component works for my new setup or if I move to another.

erikarenhill, 2025-09-26:
Hello, sorry for missing this issue. Im currently on a travel and did not upgrade to 2025.09 myself yet. I will absolutely fix this when back home in a few days